hku-mars/FAST_LIO #280: Odometry Drifted with error Integer indices would overflow

gzw1101, 2023-11-01:
Hi,

When I run the location program with mid360,everything was fine at first. but when running for about half an hour ,I got the following error:
My robot drifts out of the map and  the estimated Odometry suddenly increase to very big value as follows:

[image]

can u tell me the reason of this error and how to fix it?

Thanks.

closed 2023-11-24 by stale[bot]

eatonlu, 2026-01-05:
> Hi @gzw1101 <https://github.com/gzw1101> ,
>
> 我最近遇到了你提到的问题——长时间FAST_LIO跑（大约30分钟到1小时）后，里程表开始明显漂移或振荡。
>
> 我想知道你是否最终找到了解决这种行为的解决方案或变通方法。
>
> 如果能提供一些关于可能原因（例如IMU偏置、数值不稳定、记忆积累等）或参数调整的建议，我将非常感激。
>
> 非常感谢大家提前帮忙！
>
> me too! are you resolve?

Hi @xiaopengpeng，

我目前也還沒有完全解決這個問題，不過在測試過程中發現了一些可以延長穩定時間的方式，也分享給你參考。

從測試來看我認為這個長時間運行後的漂移／震盪的現象，可能與 iKD-Tree
隨著地圖越來越大、搜索效率逐漸下降有關。當查找變慢或不穩定時，可能會造成殘差計算不一致，最終導致里程估計發散。

一個暫時性的緩解方法，是*提高降採樣的體素大小*，讓地圖增長得慢一點。這並不能根本解決問題，但可以讓 FAST_LIO 在長時間運行時保持更久的穩定性。

在我的測試中將降採樣的大小從0.1改為0.5 可以讓穩定的時間從30分鐘延長到2個多小時

當然如果你之後找到更根本的解決方式，也歡迎在跟我分享。我這邊如果有進一步的進展

eatonlu, 2026-01-05:
> @eatonlu <https://github.com/eatonlu> hi
> 你好，您说的参数是指这几个吗，我的目前就是0.5，但是我也坚持不了两个小时。。
> 我的应用场景是用在底盘车上，就在室内进行简单的导航，
> 我还做过这样的测试，让车保持在原点不动，启动导航程序大约一个小时后，车的tf就漂移了，
> 我怀疑是雷达时间和主机时间同步的问题，因为我的地图并不大，两者存在时间误差，随着运行时间累积，误差越来越大，
> 如果可以的话，方便加个qq之类的联系方式吗，后面可以一起研究分享。
> image.png (view on web)

是的，我指的參數大致是這些。不過除了調大 surf 的降採樣外，
我目前也將 blind 設定為 0.5，在我的場景中可以明顯延長穩定運行的時間，
你可以試試看是否對你的系統也有效。

我這邊也還在持續測試，如果你有更多觀察結果，也歡迎一起交流。
如有需要進一步私下討論，也可以透過 email 聯絡。

xiaopengpeng, 2026-01-05:
> 是的，我指的參數大致是這些。不過除了調大 surf 的降採樣外，
> 我目前也將 blind 設定為 0.5，在我的場景中可以明顯延長穩定運行的時間，
> 你可以試試看是否對你的系統也有效。
> 
> 我這邊也還在持續測試，如果你有更多觀察結果，也歡迎一起交流。
> 如有需要進一步私下討論，也可以透過 email 聯絡。
> 
> xiaopengpeng ***@***.***> 於 2026年1月5日週一 上午10:28寫道：
> […](#)

我按照你的参数改了，我还额外改了livox_ros_driver2的点云发布频率，我把发布频率增大了一倍，连续运行时间似乎也增加了一倍，可以连续运行两个小时，您有时间也可以尝试下，我还有个问题不知道你遇到没，就是这个关于时间的日志打印

[component_container_isolated-6] [INFO] [1767583083.315353273]: Message Filter dropping message: frame 'livox_frame' at time 1767583082.800 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

ddinocoe, 2026-04-14:
@xiaopengpeng @eatonlu 你好我想請問一下，我也是使用Livox mid360 我的點雲發布頻率是50hz，fastlio2的參數則是為下：
```
        ros__parameters:
        feature_extract_enable: false
        point_filter_num: 3
        max_iteration: 3
        filter_size_surf: 0.5
        filter_size_map: 0.5
        cube_side_length: 1000.0
        runtime_pos_log_enable: false
        map_file_path: "./test.pcd"

        common:
            lid_topic:  "/livox/lidar"
            imu_topic:  "/livox/imu"
            time_sync_en: false     
            time_offset_lidar_to_imu: 0.0 

        preprocess:
            lidar_type: 1         
            scan_line:  4
            blind: 0.5
            timestamp_unit: 3
            scan_rate: 10
```

我的運用為室內無人機飛行，只不過fastlio2跑了大約五分鐘就會開始漂移。我有嘗試把filter_size_surf 和 filter_size_map 改至 0.8， 但是也只能延長運營時間至八分鐘，請問你們有遇到類似的問題嗎？