hku-mars/FAST_LIO #281: Odometry drift far after localization code running a period of time

wyb17js, 2023-11-01:
Hi,
we are using ros2 to realize the automatic navigation function of the car, but the following problems appear in the positioning process:

There was no problem at the beginning of the program, the car gained its initial positioning and did not lose position in the process of moving. But after about 20 minutes of running, the car could not get its current position on the map, drifted far away, and the odometer increased and increased very fast.

The dark part is the local map, the light part is the global map, and the green part is the car body，The initial positioning state is as follows:
[image]
The lost position is such that the car has flown out of view and console prompts no effective point:
[image]

We wanted to know why this happened, it is so suddenly and without warning.

Thank you!

wyb17js, 2023-11-01:
It's worth mentioning that when I increased the scanning frequency of the radar, the positioning duration was extended.
The radar we use is mid360.

ryanpdoc57, 2023-11-02:
@wyb17js are you using humble or foxy?

wyb17js, 2023-11-06:
> @wyb17js are you using humble or foxy?

humble,do you have an opinion?

closed 2023-11-29 by stale[bot]