hku-mars/FAST_LIO #282: Too large IMU bias in estimated state.

LarryDong, 2023-11-01:
Hi, 

I exported state estimated by esekf, and plotted all state variables. 
I found that the bias of IMU, e.g., `ba` drift too large that it become 3-4 m/s^2 after 300s. See the figure below:

[image]

The data is recorded by handheld avia lidar, and `mapping_avia.launch` is runned.
The map and trajectory are both correct. 
More about the figure: 1) the x-axis is "the number of scans" but not the real time, so x-tick/10 is the real time in second; 2) `ba` is already multipled by G;

Then I test a static case using mid360 and `mapping_mid360.launch`.
The lidar is placed stable without any movement, for 150s. Then the bias `ba` is also drifting to 0.8m/s^2:

[image]

I also viewed the raw IMU data, and I'm sure that the IMU does not have such severe drift. (`ba`-y is always 0.2m/s^2 because my IMU is a little tilted.)

So could anyone explain why this happens? Does the fast-lio estimate the bias wrong?

closed 2023-11-23 by LarryDong

wyb17js, 2023-12-18:
Hi bro,have you solve this problem?I meet the same question with you.This caused my position to be lost

LarryDong, 2023-12-18:
> Hi bro,have you solve this problem?I meet the same question with you.This caused my position to be lost

The drift is not so large as shown in the figure. I multiplied the output by 9.8 but it was already in m/s2 not `g`. So the drift is minor.  
Besides, I dno't think the large drift will lead to position lost. Please check your lidar data I suggest.

wyb17js, 2023-12-20:
> > Hi bro,have you solve this problem?I meet the same question with you.This caused my position to be lost
> 
> The drift is not so large as shown in the figure. I multiplied the output by 9.8 but it was already in m/s2 not `g`. So the drift is minor. Besides, I don't think the large drift will lead to position lost. Please check your lidar data I suggest.

During the operation of my program, attitude estimation error will suddenly appear, which is related to the longer utilization time of radar data after detection. For example, my radar is 10HZ, and the radar information is used once for 0.1s, but with the increase of time, there will be an interval between the radar information used. For example, I use the radar data of 9.3s at present. Supposedly next time I should use the 9.4s incoming radar data, but in fact I may be using the 9.8s radar data, and this interval will be larger and larger, when the 1s are reached, the positioning is lost.

LarryDong, 2023-12-20:
> During the operation of my program, attitude estimation error will suddenly appear, which is related to the longer utilization time of radar data after detection. For example, my radar is 10HZ, and [...]

I think your issue is not IMU bias drift. Please check your raw lidar output, the lidar may fail to work under too high/cold temperature or other communication issue.  
You may need open a new issue for this issue, not under current this issue.

wyb17js, 2023-12-21:
> I think your issue is not IMU bias drift. Please check your raw lidar output, the lidar may fail to work under too high/cold temperature or other communication issue. You may need open a new issue [...]

OK，thank you!