hku-mars/FAST_LIO #286: 测试rosbag包时无法显示点云

x710777335, 2023-12-05:
你好，我在运行rosbag测试的时候，程序出现QObject::connect: Cannot queue arguments of type 'QVector<int>'(Make sure 'QVector<int>' is registered using qRegisterMetaType().)错误，rviz无法显示点云图像，我看有人遇到过这种现象，但没说怎么解决。

[ INFO] [1701769842.323935065]: IMU Initial Done
[ WARN] [1701769842.324732143]: No point, skip this scan!

QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
^C[rviz-3] killing on exit
[laserMapping-2] killing on exit
[ WARN] [1701769987.536904851]: catch sig 2
current scan saved to /PCD/scans.pcd
^C^C^CRebuild thread terminated normally
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

closed 2023-12-28 by stale[bot]

shaoxiang, 2024-07-09:
> sorry, you fix this question anymore? I also want to know how to do it
Refer to #278, please check the timestamps of IMU and Lidar

cuboya, 2024-11-21:
您好，请问一下怎么解决的呢，我也有这个问题

x710777335, 2024-11-24:
当时没有解决这个问题，应该是点云数据格式的问题，fast-lio里面点云有2种格式，注意看下配置文件

十一
***@***.***

cuboya, 2024-11-24:
> 当时没有解决这个问题，应该是点云数据格式的问题，fast-lio里面点云有2种格式，注意看下配置文件 十一 ***@***.***  
> […](#)

很高兴收到您的答复，我想问一下
请问是下面这段吗

preprocess:
    lidar_type: 1                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line:  6
    blind: 4

我是使用的提供的bag包
https://drive.google.com/drive/folders/1CGYEJ9-wWjr8INyan6q1BZz_5VtGB-fP
请问是用哪个格式呢

FubianTiger, 2025-03-16:
> > 当时没有解决这个问题，应该是点云数据格式的问题，fast-lio里面点云有2种格式，注意看下配置文件 十一 _**@**_.***  
> > […](#)
> 
> 很高兴收到您的答复，我想问一下 请问是下面这段吗
> 
> preprocess: lidar_type: 1 # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, scan_line: 6 blind: 4
> 
> 我是使用的提供的bag包 https://drive.google.com/drive/folders/1CGYEJ9-wWjr8INyan6q1BZz_5VtGB-fP 请问是用哪个格式呢

您好，当时我遇到了同样的问题，倘若您使用的是VMware/Ubuntu18.04搭建的ROS，可以参考这个方法：
https://blog.csdn.net/Pichai/article/details/123648573