hku-mars/FAST_LIO #291: Indoor odometers are not very accurate

yangisod, 2023-12-28:
Indoor odometers are not very accurate.
Inside I rotated the lidar around and the walls had a very obvious multiple layers
At the same time, the terminal keeps reporting error voxel leaf size. I tried to increase filter size surf, but no error was reported until 0.4
What parameters should I debug for this kind of problem

[image]
[image]

yangisod, 2023-12-28:
[image]
I ran the scan again and the walls were still quickly stratified, what should I do

closed 2024-01-26 by stale[bot]

dom-lee, 2024-02-18:
It would help if you tried to decrease `blind` parameter