hku-mars/FAST_LIO #304: Downsampling Error

Leeseunghun03, 2024-02-13:
Hello
I'm trying to test the code with VLP 16, but I'm having a problem.
```
downSizeFilterSurf.setInputCloud(feats_undistort);
downSizeFilterSurf.filter(*feats_down_body);
t1 = omp_get_wtime();
feats_down_size = feats_down_body->points.size();
/*** initialize the map kdtree ***/
if(ikdtree.Root_Node == nullptr)
{
                if(feats_down_size > 5)
                {
                    ikdtree.set_downsample_param(filter_size_map_min);
                    feats_down_world->resize(feats_down_size);
                    for(int i = 0; i < feats_down_size; i++)
                    {
                        pointBodyToWorld(&(feats_down_body->points[i]), &(feats_down_world->points[i]));
                    }
                    ikdtree.Build(feats_down_world->points);
                }
                continue;
}
```
In the process of downsampling in the above code, the filtered point cloud size(feats_down_size) is always 1, so it continues and the code cannot proceed indefinitely. 
Based on filter_size_surf of 0.5 and point_filter_num of 4, the input point cloud size is about 6000, and the input value seems to be normal. 
Why does it cause data loss during the downsampling process? Am I missing something here? 
Thanks!

closed 2024-03-07 by stale[bot]

willat343, 2024-09-04:
I had the same issue. Input LiDAR data was fine, but after being processed by `p_imu`, the cloud was mostly NaN values. This is why the voxelgrid only produces a small number of points after filtering, since it gets rid of the NaN points. I still don't know why the IMU processing unit does this, perhaps some issue with the IMU data, or the pointcloud timestamps.

xchwang1998, 2025-05-24:
I have same issue. Is there effective solution now?