hku-mars/FAST_LIO #308: Gmapping navigation

fightforge, 2024-02-26:
Hi,
I'm trying to generate a map using slam_gmapping but i cannot get it to work, I'm a newbie in ros and I don't know if it's the best way to use gmapping to obtain 2d point and click navigation on rviz.
I'm using a livox mid-360 lidar on a 4wd direct drive rover controlled by roboclaw controller but without encoders. I'm using your code to get odometry and pointcloud_to_laserscan to convert pcl to 2d laserscan but i cannot get slam_gmapping to work.
Could you suggest me another way to get 2s point and click navigation using your code?

Thank you

proxi666, 2024-03-05:
Hey I am also working with Livox MID-360, to make an large scale AMR..
I have successfully created a map using this algorithm, do you want to collaborate and work together in the project?

ubicray, 2024-03-06:
Any chance you guys wanna share your config files for mid360? Don't see one anywhere around here

fightforge, 2024-03-18:
Cannot collaborate, I'm not a developer.  For this issue, I'm trying point_lio, it seems to work better

closed 2024-04-15 by stale[bot]

bcodesnow, 2024-07-05:
@fightforge You seemed to have success with pointcloud_to_laserscan and the livox mid 360. I am facing issues with it.
ROS (1) Noetic - did you use ROS2? 
https://github.com/ros-perception/pointcloud_to_laserscan/issues/91

The same install is working without issues with a realsense rgbd camera. 
Any idea is welcome! Thank you in advance.