hku-mars/FAST_LIO #313: Is IMU optional?

ngtroku, 2024-03-18:
Does this SLAM method require IMU topic? I am interested in evaluating the impact of the presence or absence of an IMU (Inertial Measurement Unit) on map generation.

zhh2005757, 2024-03-23:
> Hi, thank you for your great work. Does this SLAM method require IMU topic? I am interested in evaluating the impact of the presence or absence of an IMU on map generation.

Maybe you can refer to their Lidar-IMU calibration project "LI-Init". They develop a LO named FAST-LO based FAST-LIO  framework.

ngtroku, 2024-03-26:
@zhh2005757 
Thank you for your response. I would like to know if FAST-LIO can work without an IMU.

closed 2024-04-20 by stale[bot]

renatozapatalusni, 2024-07-21:
@zhh2005757  I would like to know if this would be possible, did you find an answer about using FAST_LIO without the IMU?

zhh2005757, 2024-07-23:
> @zhh2005757 
> Thank you for your response. I would like to know if FAST-LIO can work without an IMU. 

I mean this repo https://github.com/hku-mars/LiDAR_IMU_Init, it develops an iekf based pure LiDAR odometry FAST-LO, which I think it is a FAST-LIO version without IMU indeed. I will extract such FAST-LO module individually in my github soon.