hku-mars/FAST_LIO #318: process has died exit code 11

rhobot95, 2024-04-09:
Hi. I'm using ouster os1-64. When I use my recorded bag file (about indoor data), I face this problem
[laserMapping-1] process has died [pid 3828, exit code -11, cmd /home/herolidarsbc/catkin_ws/devel/lib/fast_lio/fastlio_mapping __name:=laserMapping __log:=/home/herolidarsbc/ .ros/log/37b095ce-ef56-11ee-9f48-6b466cc69d5c/laserMapping -1.log].

Also, I wonder why the warning "Failed to find match for field 'ambient'" occurred. Is field 'ambient' about livox lidar?

[image]

closed 2024-05-03 by stale[bot]

Vor-Art, 2024-05-17:
This help me to handle this warning: https://github.com/hku-mars/LiDAR_IMU_Init/issues/73#issuecomment-2118186378

autel-liuxiaozi, 2024-07-01:
> Any knowledge on why this is happening? I have the same issue

I have same issue, any idea?

jqkzero, 2025-02-27:
> 解决了，源代码没有这个问题。但我们是修改过的代码所以有问题。将pcd点云输出到了其他路径，然后却又没有创建那个目录，导致出错

你好，请问具体是怎么解决的呢