hku-mars/FAST_LIO #323: Memory leak

SabraHashemi, 2024-04-29:
I have experienced a huge memory leak by running this repo, i have set  pcd_save_en flag in yaml file to false, but the problem persisted 
@Ecstasy-EC

SabraHashemi, 2024-04-29:
I’ve detected that its cause is something related to rviz visualization but I don’t know how to solve it

Ecstasy-EC, 2024-04-30:
Does the memory leak appear when you close the rviz? The rviz will accumulate point clouds for visualization thus increasing the memory consumption.

SabraHashemi, 2024-04-30:
If i use rviz:=false , the memory leakage slows down, but still exist and happens
@Ecstasy-EC

Ecstasy-EC, 2024-05-04:
> @Ecstasy-EC I am waiting for your response, this issue is really important I guess , I have noticed that some people had the same issue before, your code is working really good but the memory leak prevents us to enjoy it :)

Hi @SabraHashemi. Sorry for my late reply. Would you please provide me with the memory-time plot to see how fast it increases? And is the odometry working normally? Besides, the last person who met this problem solved it by updating with the latest commit in this repo. Did you check that everything was on the latest version?

SabraHashemi, 2024-05-04:
@Ecstasy-EC actually I’ve got the code recently ( 2 weeks ago) so I’m using your latest version and I’m using melodic Ros, odometry is working normally 
About memory time plot, i will provide it soon , but to know more about my issue
I’m using a 16 g Ram , and with a data from velodyne in the repo, and i can see that every 2 sec memory usage will increased about 0.01 gig

[image]

Ecstasy-EC, 2024-05-04:
> @Ecstasy-EC actually I’ve got the code recently ( 2 weeks ago) so I’m using your latest version and I’m using melodic Ros, odometry is working normally About memory time plot, i will provide it soon , but to know more about my issue I’m using a 16 g Ram , and with a data from velodyne in the repo, and i can see that every 2 sec memory usage will increased about 0.01 gig
> 
> ![IMG_6736](https://private-user-images.githubusercontent.com/17723386/327954227-c375a04a-416d-49a1-bd4d-4f2b3c17293f.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ4MzI0NDYsIm5iZiI6MTcxNDgzMjE0NiwicGF0aCI6Ii8xNzcyMzM4Ni8zMjc5NTQyMjctYzM3NWEwNGEtNDE2ZC00OWExLWJkNGQtNGYyYjNjMTcyOTNmLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNFQxNDE1NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMDhmY2Y2OTY2MTQwYmU0OTkzZjE2ZTM1NDk0MmY4MzBlZDc3ZDVmZGI5ZTljN2MxZGRhY2I1MmFiZDg2OTcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.lDhbdiUrC-zVY_8EeS-wEU6J6UPld9EMmFTehhBTz_c)

Quite weird. Could you please also provide me with a rosbag and a config file in your test? Which type of LiDAR do you use?

SabraHashemi, 2024-05-04:
I’m using velodyne HE 32 data you have provided in the repo ( the ine which is about 22 gig) 
And your velodyne.yaml file , the only differences is that i have changed lidar topic to : points_raw 
The imu topic: imu_raw 
Save pcd to false 
@Ecstasy-EC

SabraHashemi, 2024-05-05:
I know that you are busy, so consider this as a kind reminder @Ecstasy-EC

Ecstasy-EC, 2024-05-05:
> I know that you are busy, so consider this as a kind reminder @Ecstasy-EC 

Could you provide your rosbag for me to have a look?

SabraHashemi, 2024-05-05:
@Ecstasy-EC sorry, 
https://drive.google.com/drive/folders/1blQJuAB4S80NwZmpM6oALyHWvBljPSOE?usp=sharing

The 20 gig bag

Ecstasy-EC, 2024-05-06:
> @Ecstasy-EC sorry, https://drive.google.com/drive/folders/1blQJuAB4S80NwZmpM6oALyHWvBljPSOE?usp=sharing
> 
> The 20 gig bag

I will look into it but please patiently wait for several days.

AdamBarghati, 2024-05-29:
Hi @Ecstasy-EC 
Any updates on this issue?

closed 2024-06-23 by stale[bot]