hku-mars/FAST_LIO #337: How to run FAST_LIO by using Livox and external IMU?

WN-Wolf, 2024-07-10:
I am a beginner, if you by pass, please give me some advice, many thanks!
lidar：livox mid-70
IMU：wit JY901B

Q1: If I get a LiDAR_IMU_Init result, like this:
I want to know, how to use this result in FAST_LIO? Where to use this result?
[image]

Q2: For Livox serials with external IMU
I have modified config/avia.yaml like this.
[image]
and then run command, "roslaunch fast_lio mapping_avia.launch".

What about the next? run this command "rosbag play my_record_lidar_imu.bag"? 

How to get "my_record_lidar_imu.bag"? 
Is it recorded using this command after time synchronization? 
"rosbag record -o my_livox_wit.bag /livox/lidar /wit/imu"

WN-Wolf, 2024-07-12:
Let me ask my question! haha!
1. How to use the Initialization_result.txt.
[image]
2. How to run fast_lio2
first terminal: roslaunch livox_ros_driver livox_lidar_msg.launch
second terminal: roslaunch wit_ros_imu wit_imu.launch
last terminal: roslaunch fast_lio mapping_avia.launch
pcd file path: /fast_lio2/src/FAST_LIO/PCD, then download pcl_viewer to watch pointclound.

closed 2024-08-01 by WN-Wolf

jongco22, 2024-10-01:
Thank you for your question ^^*

fitzzzzzzzzz, 2024-12-17:
hi @nw-wf , i am also beginner, could you please tell me how did you get the initilization.txt result? 
i am using livox mid 360

nw-wf, 2024-12-19:
> hi @nw-wf , i am also beginner, could you please tell me how did you get the initilization.txt result? i am using livox mid 360

I have gave up using external lidar and I have bought mid-70 to run fastlio.

fitzzzzzzzzz, 2024-12-20:
oh :/ 
I am also using mid360 now, but if I rotate two times in the same place, the graph drifts in height and position a little bit. it does not exactly match. i guess loop closure isn't working properly.

fitzzzzzzzzz, 2024-12-20:
also can you please tell me though, how did you run this initialization script @nw-wf ? 
how are you getting these values of Initialization_result?

nwf-oo, 2025-02-13:
> also can you please tell me though, how did you run this initialization script @nw-wf ? how are you getting these values of Initialization_result?

please to refer to this https://github.com/hku-mars/LiDAR_IMU_Init/issues/108#issue-2397438510