hku-mars/FAST_LIO #338: Lidar rotation isn't working

er1c-cartman, 2024-07-12:
Hi,

I have been using your excellent work, but unfortunately, I am encountering a serious issue.

To use fast_lio in my robot, I have rotated the LiDAR 180 degrees on the x-axis. However, the map appears upside down; the LiDAR point cloud is pointing at the ceiling instead of the floor.

I have modified the URDF to apply the rotation on the x-axis, but it hasn't resolved the issue.

How can I fix this error?

[image]

er1c-cartman, 2024-07-12:
For better understanding, I'm uploading more images

[image]

[image]

And i have made a turn on xacro

  <xacro:include filename="$(find ros2_livox_simulation)/urdf/mid360.xacro" />
  <xacro:mid360 name="livox_frame" parent="base_link" topic="/livox/lidar">
    <origin xyz="0 0 0.86521" rpy="3.14 0 0"/>
  </xacro:mid360>

IacopomC, 2024-07-31:
I think this issue may help you #284 althought it doesn't fully solves the problem (https://github.com/hku-mars/FAST_LIO/issues/284)

closed 2024-08-24 by stale[bot]