hku-mars/FAST_LIO #339: Lidar pointcloud just 1 straight line

RBinnenmars, 2024-07-12:
I have a problem with trying to use Fast-lio with the following sensors
IMU: Xsense MTI-300
Lidar: Hesai Pandar XT-32

Because the repo did not have a way to handle the Pointcloud messages from the Hesai pandar, a Pander handler function was added to the preprocess.cpp and .h files, which was retrieved from another issue for the repo that also uses . After setting everything up and determining the rotation between lidar and imu, there were some issues. The sensors were not synchronized, after adding a "time_offset_lidar_to_imu" a point cloud was visualized, which can be seen in the figure below. The point cloud is just several individual lines and not a rectangular room as was measured, hope someone can help me with this.

The possible sources, that I can think of, for this error, are: Time synchronization, extrinsic transformation, or a coding error. 

common:
    lid_topic:  "/hesai/pandar"
    imu_topic:  "/imu/data/"  
    time_sync_en: false
    time_offset_lidar_to_imu: 130253573.95681909 

preprocess:
    lidar_type: 4                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
    scan_line: 32
    timestamp_unit: 3          # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
    blind: 3

mapping:
    acc_cov: 0.1
    gyr_cov: 0.1
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    fov_degree:    360
    det_range:     120.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic
    extrinsic_T: [ 0,0.075,-0.075]
    extrinsic_R: [ 0,  -1,  0 
                   0,   0,  1,
                  -1,   0,  0]

publish:
    path_en:  false
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1              

[image]

RBinnenmars, 2024-07-12:
I saw that a comma was missing in the extrinsic_R variable, now the point cloud looks like the this but the lines are still small, anyone any ideas.

[image]

RBinnenmars, 2024-07-17:
this is/was a timing issue, recorded another bag file where the time was synchronized and the point cloud was much better

closed 2024-07-17 by RBinnenmars

85256638, 2024-12-05:
@RBinnenmars can you kindly share how you sync your Hesai Pandar XT-32 and IMU?