hku-mars/FAST_LIO #347: Failed in marsim simulation

Yangrlan, 2024-08-06:
Hi authors:
The fast-lio2 drifts when the drone begins to move in the marsim simulator. The planner I used is ego-planner and the trajectory is not aggressive. I changed to use mid360 but it still drifts. It puzzles me for a long time as I have tried before the fast-lio2 supports the simulator. And may you give me some suggestions?

ngocquynhle, 2024-08-28:
Hi @Yangrlan 
Have you solved this problem yet?

closed 2024-09-21 by stale[bot]

Yangrlan, 2024-09-27:
I found the problem today. Maybe in the current simulator, the topic of the pointcloud that fast-lio should use is "/robot_0/velodyne_points" but not the "/quad0_pcl_render_node/cloud". Because the frame of the previous one is body frame and the latter one is world frame.