hku-mars/FAST_LIO #348: Don't save map

Parkeunseok, 2024-08-06:
It is currently in use by ubuntu 22.04 ros2 humble

The config file I set is as follows.

As you can see in the file below, the map is not saved even though "pcd_save_en" is true.

[image]

[image]

What is the solution?

Ericsii, 2024-08-12:
In order not to be affected by the io operation of the saved map performance, I made a new ros service `map_save` with type std_srvs::srv::Trigger. You can try manually trigger the service to save the map pointcloud to file.

Parkeunseok, 2024-08-14:
i'm solved it thank you

123295, 2024-08-14:
I've meet the same problem and I can't save the pcd file, could you show me your folder structure? Thank you!

pk30123, 2024-09-02:
I tried executing the service map_save, but i did not give any results , Can you please help on elaborating it

pk30123, 2024-09-02:
[fastlio_mapping-1] [INFO] [1725266567.425343949] [laser_mapping]: Saving map to /home/pradheep/Documents/flio_ws/src/FAST_LIO/PCD/test.pcd...
[fastlio_mapping-1] terminate called after throwing an instance of 'pcl::IOException'
[fastlio_mapping-1]   what():  : [pcl::PCDWriter::writeBinary] Input point cloud has no data!
[ERROR] [fastlio_mapping-1]: process has died [pid 5983, exit code -6, cmd '/home/pradheep/Documents/flio_ws/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/pradheep/Documents/flio_ws/src/FAST_LIO/config/ouster64.yaml --params-file /tmp/launch_params_4__njwjw'].

arjunskumar, 2024-09-02:
@pk30123 check the size of pointcloud you are trying to save

pk30123, 2024-09-02:
Do you know which topic i should check for the point cloud? i could see rviz building. Same dataset in ros1 docker works

arjunskumar, 2024-09-02:
please check the directory in which map is saving? @pk30123

pk30123, 2024-09-02:
Hi, sorry to elaborate i used docker in FASTLIO ROS1 and FASTLIO2 . the calibration matrices are the same , i could see some output in RVIZ in both but there is no saved pcd files in ROS2

lidarmansiwon, 2024-09-12:
@pk30123 I had the same problem and error. I tried various ways to save the map.
My environment is humble and the solution was to uncomment the code below and the map was saved.

laserMapping.cpp

Uncommented the comment code at line 513.

/**************** save map ****************/
/* 1. make sure you have enough memories
/* 2. noted that pcd save will influence the real-time performences **/
if (pcd_save_en)
{
int size = feats_undistort->points.size();
PointCloudXYZI::Ptr laserCloudWorld( \
new PointCloudXYZI(size, 1));

for (int i = 0; i < size; i++)
{
RGBpointBodyToWorld(&feats_undistort->points[i], \
&laserCloudWorld->points[i]); } *pcl_wait_save += *laserCloudWorld;

 static int scan_wait_num = 0;
 scan_wait_num++;
 if (pcl_wait_save->size() > 0 && pcd_save_interval > 0 && scan_wait_num >= pcd_save_interval) { pcd_index ++;
 string all_points_dir(string(string(ROOT_DIR) + "PCD/scans_") + to_string(pcd_index) + string(".pcd"));
 pcl::PCDWriter pcd_writer;
 cout << "current scan saved to /PCD/" << all_points_dir << endl;
 pcd_writer.writeBinary(all_points_dir, *pcl_wait_save);
 pcl_wait_save->clear();
 scan_wait_num = 0;
 }
}
}

pk30123, 2024-09-16:
Hi,

Thanks for the update. This made the map to save on Ctrl+C not on srv .

Thanks

closed 2024-11-09 by stale[bot]

G7-3208, 2025-04-23:
> [fastlio_mapping-1][信息][1725266567.425343949] [laser_mapping]：正在将地图保存到 /home/pradheep/Documents/flio_ws/src/FAST_LIO/PCD/test.pcd... [fastlio_mapping-1] 在抛出 'pcl：：IOException' 实例后终止调用 [fastlio_mapping-1] what（）： ： [pcl：:P CDWriter：：writeBinary] 输入点云没有数据！ [错误] [fastlio_mapping-1]：进程已死机 [pid 5983，退出代码 -6，cmd '/home/pradheep/Documents/flio_ws/install/fast_lio/lib/fast_lio/fastlio_mapping --ros-args --params-file /home/pradheep/Documents/flio_ws/src/FAST_LIO/config/ouster64.yaml --params-file /tmp/launch_params_4__njwjw']。

Hello, I have encountered the same problem as you. How did you solve it

lesurJ, 2026-01-20:
same error for me