hku-mars/FAST_LIO #373: Hi bag from ros1 to ros2 continue in type of livox ros driver

hello299, 2024-11-27:
您好。 我目前遇到一个问题，通过利用rosbags-converter 我发现对应的bag中的/livox/lidar信息仍然是livox_ros_driver类型，导致我在程序中利用livox_ros2_driver或者livox_ros_driver2都不能正确读取点云数据。 我的环境是ubuntu 22.04。
我已经将我的程序适配到了livox_ros2_driver（适配avia的版本），但是使用 [https://github.com/hku-mars/FAST_LIO ](https://github.com/Ericsii/FAST_LIO_ROS2/issues/url)提供的数据包，经过转换的db3数据类型仍然是livox_ros_driver,我改完程序的消息接收端口适配的是livox_ interface，如下图，请问您能够提供相关的改进建议么？
[image]

closed 2024-12-28 by stale[bot]

Yc-Dlan, 2025-09-03:
Hello @illusionaryshelter  I modified the data type in YAML as you suggested, but it doesn't seem to work.
```bash
rosbag2_bagfile_information:
  compression_format: ''
  compression_mode: ''
  custom_data: null
  duration:
    nanoseconds: 351179631650
  files:
  - duration:
      nanoseconds: 351179631650
    message_count: 106020
    path: 11.db3
    starting_time:
      nanoseconds_since_epoch: 1612525854719196537
  message_count: 106020
  relative_file_paths:
  - 11.db3
  ros_distro: rosbags
  starting_time:
    nanoseconds_since_epoch: 1612525854719196537
  storage_identifier: sqlite3
  topics_with_message_count:
  - message_count: 70901
    topic_metadata:
      name: /livox/imu
      offered_qos_profiles: ''
      serialization_format: cdr
      type: sensor_msgs/msg/Imu
      type_description_hash: RIHS01_7d9a00ff131080897a5ec7e26e315954b8eae3353c3f995c55faf71574000b5b
  - message_count: 35119
    topic_metadata:
      name: /livox/lidar
      offered_qos_profiles: ''
      serialization_format: cdr
      type: livox_ros_driver2/msg/CustomMsg
      type_description_hash: RIHS01_94041b4794f52c1d81def2989107fc898a62dacb7a39d5dbe80d4b55e538bf6d
  version: 8
```
```bash
yc-dlan@fishros-linux:~/fast_lio2_dlan/ros2_ws/src/FAST_LIO/bag/ros2bag$ ros2 bag play 11.db3 

closing.

closing.
[INFO] [1756908315.248320664] [rosbag2_storage]: Opened database '11.db3' for READ_ONLY.
[INFO] [1756908317.324121586] [rosbag2_player]: Set rate to 1
[WARN] [1756908317.325175962] [rosbag2_player]: Ignoring a topic '/livox/lidar', reason: package 'livox_ros_driver' not found, searching: [/home/yc-dlan/packages/ws_livox/install/livox_ros_driver2, /home/yc-dlan/fast_lio2_dlan/ros2_ws/install/fast_lio, /opt/ros/humble].
[INFO] [1756908317.326335396] [rosbag2_player]: Adding keyboard callbacks.
[INFO] [1756908317.326356275] [rosbag2_player]: Press SPACE for Pause/Resume
[INFO] [1756908317.326365904] [rosbag2_player]: Press CURSOR_RIGHT for Play Next Message
[INFO] [1756908317.326374289] [rosbag2_player]: Press CURSOR_UP for Increase Rate 10%
[INFO] [1756908317.326381102] [rosbag2_player]: Press CURSOR_DOWN for Decrease Rate 10%

closing.

closing.
[INFO] [1756908317.326736462] [rosbag2_storage]: Opened database '11.db3' for READ_ONLY.
```

huang98778, 2026-10-08:
[image].  播放ros2的文件夹貌似可以