hku-mars/FAST_LIO #388: 长距离建图，掉头位置建图失败

MengHuiLei, 2025-02-11:
您好：

nessonma, 2025-02-11:
@XW-HKU 补充一下，同一个掉头，我从掉头位置起步或距掉头位置不远起步（比如说500m），然后再进行掉头都没有问题，貌似超过1km后，掉头位置必然产生异常

MengHuiLei, 2025-02-11:
[image]

Huguet57, 2025-02-12:
This should be the habitual experience when trying long paths. This repository does not have loop closure so the drift error keeps propagating. Maybe you would be interested in Loop Closure repositories such as: https://github.com/yanliang-wang/FAST_LIO_LC

MengHuiLei, 2025-02-13:
> This should be the habitual experience

试了，没起作用。猜测没起作用的原因是：回环检测无法消除匹配不好的情况

Huguet57, 2025-02-17:
Where does the drift get worse? Maybe if it's in a very quick turn you might want to check your IMU is well-calibrated and/or the axis are well placed. Try only integrating the IMU to see if this makes sense. Try other GitHub repositories and see if it also happens with other algorithms (you can try [LIMO](https://github.com/Huguet57/LIMO-Velo)).

closed 2025-03-15 by stale[bot]