hku-mars/FAST_LIO #389: ROS2 Humble build issue

marapakabannudeepak, 2025-02-20:
CMake Error at CMakeLists.txt:95 (add_executable):
  Cannot find source file:

    include/ikd-Tree/ikd_Tree.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at CMakeLists.txt:95 (add_executable):
  No SOURCES given to target: fastlio_mapping
There is no such folder include
add_executable(fastlio_mapping /home/inertia/ws_1/src/FAST_LIO-ROS2/src/laserMapping.cpp include/ikd-Tree/ikd_Tree.cpp src/preprocess.cpp)

admin-init, 2025-02-20:
I have encountered the same issue when attempting to compile the [FAST_LIO_ROS2](https://github.com/Ericsii/FAST_LIO_ROS2). 

I'm not sure if the problem has been that I download the source code by .zip rather than `git clone`.

### Here's what worked for me:

According to the CMake Error, I checked the `FAST_LIO_ROS2-ros2/include/ikd-Tree` which is empty. Then I check the github repositor where links to [ikd-Tree](https://github.com/hku-mars/ikd-Tree/tree/e2e3f4e9d3b95a9e66b1ba83dc98d4a05ed8a3c4).

Therefore, I have tried to download these files `ikd_Tree.cpp  ikd_Tree.h  README.md` into my local  `FAST_LIO_ROS2-ros2/include/ikd-Tree` which seemed fix this complile problem.

stergiosba, 2025-02-28:
You do not have to download any files manually. Git is already a very nicely made tool.
[ikd-Tree @ e2e3f4e](https://github.com/hku-mars/ikd-Tree/tree/e2e3f4e9d3b95a9e66b1ba83dc98d4a05ed8a3c4) is added as a submodule to this repo.
Clone the repo as:
```bash
git clone --recurse-submodules https://github.com/hku-mars/FAST_LIO.git
```
Notice the ` --recurse-submodules` flag will populate the empty folder you are referring to.

closed 2025-03-23 by stale[bot]