hku-mars/FAST_LIO #394: mid360运行fastlio的里程计发布频率问题

NarcissusEcho, 2025-03-08:
您好，使用mid360跑fastlio的里程计发布频率只有10hz，对于无人机应用来讲太低了，修改livox_ros_driver中的publish_freq参数又会导致里程计数据严重失真。
请问如何较好的将fast lio的里程计发布频率提高

teacher1314, 2025-03-20:
我也想知道这个答案，的确发布频率太低了

youwyu, 2025-03-26:
You can publish the odometry from IMU integration at IMU_preprocessing. If you refer to the code "imu_state = kf_state.get_x();", you can write a Odom Publisher at class ImuProcess, and publish "imu_state".

teacher1314, 2025-03-27:
> You can publish the odometry from IMU integration at IMU_preprocessing.hpp. If you refer to the code "imu_state = kf_state.get_x();", you can write a Odom Publisher at class ImuProcess, and publish "imu_state". Note: the "imu_state = kf_state.get_x();" code is in the for loop under code UndistortPcl.

好的，谢谢你的回答，我之前已经找到如何来实现高频里程计了，正如你所说的那样

dddyyyy123, 2025-04-14:
> > 您可以在 IMU_preprocessing.hpp 上发布来自 IMU 集成的里程计。如果您引用代码 “imu_state = kf_state.get_x（）;”，您可以在 ImuProcess 类中编写一个 Odom Publisher，并发布 “imu_state”。注意：“imu_state = kf_state.get_x（）;” 代码位于代码 UndistortPcl 下的 for 循环中。
> 
> 好的，谢谢你的回答，我之前已经找到如何来实现高频里程计了，正如你所说的那样

请问该怎么提高呢，而且我的速度坐标轴不会随着我飞机的自旋而旋转，这对替换飞机速度很致命。

teacher1314, 2025-04-14:
> > > 您可以在 IMU_preprocessing.hpp 上发布来自 IMU 集成的里程计。如果您引用代码 “imu_state = kf_state.get_x（）;”，您可以在 ImuProcess 类中编写一个 Odom Publisher，并发布 “imu_state”。注意：“imu_state = kf_state.get_x（）;” 代码位于代码 UndistortPcl 下的 for 循环中。
> > 
> > 
> > 好的，谢谢你的回答，我之前已经找到如何来实现高频里程计了，正如你所说的那样
> 
> 请问该怎么提高呢，而且我的速度坐标轴不会随着我飞机的自旋而旋转，这对替换飞机速度很致命。

其实就是坐标变换吧，将全局坐标系转换为自身坐标系即可

Addog666, 2025-04-23:
请问您mid360装的驱动是livox_ros_driver2吗？我看fast-lio只支持livox_ros_driver，您是怎么处理的呢？

teacher1314, 2025-04-23:
livox_ros_driver2是驱动雷达的呀，和fast_lio没关系吧？mid360得用2才能驱动

3176789950
***@***.***

dddyyyy123, 2025-04-23:
很简单，在他后面加个2就可以了

NarcissusEcho, 2025-05-08:
> 您可以在 IMU_preprocessing.hpp 上发布来自 IMU 集成的里程计。如果您引用代码 “imu_state = kf_state.get_x（）;”，您可以在 ImuProcess 类中编写一个 Odom Publisher，并发布 “imu_state”。注意：“imu_state = kf_state.get_x（）;” 代码位于代码 UndistortPcl 下的 for 循环中。

IMU集成的里程计有使用激光雷达的数据吗

YiyangMa8, 2025-05-24:
> You can publish the odometry from IMU integration at IMU_preprocessing.hpp. If you refer to the code "imu_state = kf_state.get_x();", you can write a Odom Publisher at class ImuProcess, and publish [...]

Hello, may I ask if this IMU data can be directly sent as a high-frequency odometer? Has a LiDAR odometer been used to calibrate the IMU? Or is it just a simple forward integration of IMU data, which can accumulate errors over time?

youwyu, 2025-05-24:
I have tried this, and it overall works well. It's a simple integration, but only between LiDAR frames, which means the accumulated error is within 0.1 seconds (assume LiDAR has 10 Hz).

YiyangMa8, 2025-05-25:
> I have tried this, and it overall works well. It's a simple integration, but only between LiDAR frames, which means the accumulated error is within 0.1 seconds (assume LiDAR has 10 Hz).

I'm so grateful for you reply.That is to say, using IMU as the high-frequency odometry output between two LiDAR frames results in very small errors. And when the next LiDAR frame (calculated by the next LiDAR Odometry) is received, that is, when the optimal estimate is obtained, this optimal estimate will be used as the initial value for the forward integration of the next IMU. As stated in the relevant formula in the paper, it is equivalent to calibrating the IMU throughout the entire process. Because it is a tightly coupled system, I can completely output the IMU information at the position you mentioned above as the odometry information for the entire process of positioning, right?Finally, I sincerely thank you again for your reply

Jamin-Yang, 2025-05-30:
it works for me. THANK YOU! : ）

yhzCode11, 2025-06-17:
你好，我按照上面说的输出了更高频率的imu_state，但是为什么输出的位姿轨迹是这样断断续续的？感谢

yhzCode11, 2025-06-17:
[image]
你好，我按照上面说的方法publish了imu_state，但是为什么显示的轨迹是断断续续的？

xiaotiantiian, 2025-06-19:
You can consider using the method of segmenting lidar frames to increase its frequency, which can be referred to in the paper sr_io++ for details.

youwyu, 2025-07-10:
> [image] 你好，我按照上面说的方法publish了imu_state，但是为什么显示的轨迹是断断续续的？

Seems like your IMU frame is not aligned with LiDAR. Please correct me if I'm wrong.