hku-mars/FAST_LIO #396: change angle of the 3d map bieng generated

hermanumrao, 2025-03-08:
So I am making a 3d map for some project, but I want to capture the floor too.
I have a Livox Mid 360 lidar.
So i just tilted the mount by an angle of around 40 deg about y axis.

Is there a way that I can rotate the live cloud points map being generated.
I want to use this 3d map to generate a 2d map for helping me with autonomous navigation even while mapping. (any tips for this also will be appreciated)

Thanks a ton in advance.

youwyu, 2025-03-26:
Add a static transform from world to camera_init with (0 0 0 0 40/180*PI 0).

closed 2025-04-20 by stale[bot]

KevinJeulink1, 2025-06-10:
I see, however, what would be a more permanent solution, lets say, a change in the code, such that the laser map topic publishes it already transformed. I am using ros2 humble by the way with a livox mid 360.