hku-mars/FAST_LIO #398: Issue with Ouster OS1-32 – Failed to Find Match for Field 'ring'

CrypticInfusion, 2025-03-19:
Hello,

I'm new to working with LiDAR and FAST-LIO, and I'm trying to use FAST-LIO with an Ouster OS1-32, but I keep running into issues. To set up the sensor, I copied the ouster64.yaml file and modified it to create ouster32.yaml, adjusting the necessary parameters as seen below:

```
        lid_topic: "/ouster/points"
        imu_topic: "/ouster/imu"

        //.....

        scan_line: 32
``` 

As for the **point_type** used in the driver_params.yaml from ouster-ros, I started with using 'original'.

```
    # point_type[optional]: choose from: {original, native, xyz, xyzi, xyzir}
    # Here is a breif description of each option:
    #  - original: This uses the original point representation ouster_ros::Point
    #    of the ouster-ros driver.
    #  - native: directly maps all fields as published by the sensor to an
    #    equivalent point cloud representation with the additon of ring
    #    and timestamp fields.
    #  - xyz: the simplest point type, only has {x, y, z}
    #  - xyzi: same as xyz point type but adds intensity (signal) field. this
    #    type is not compatible with the low data profile.
    #  - xyzir: same as xyzi type but adds ring (channel) field.
    #    this type is same as Velodyne point cloud type
    #    this type is not compatible with the low data profile.
    # for more details about the fields of each point type and their data refer
    # to the following header files:
    # - ouster_ros/os_point.h
    # - ouster_ros/sensor_point_types.h
    # - ouster_ros/common_point_types.h.
    point_type: original
``` 

However, every time I try to launch FAST-LIO, I get the following error:

```
[fastlio_mapping-1] Failed to find match for field 'ring'.
``` 

I've also tried setting the **point_type** to 'native' since it should have all the required fields, but I would get the following errors:

``` 
[fastlio_mapping-1] Failed to find match for field 'intensity'.
[fastlio_mapping-1] Failed to find match for field 'ring'.
[fastlio_mapping-1] Failed to find match for field 'ambient'.
``` 

I'm not sure if this issue is related to the configuration files or something else, but any help would be greatly appreciated. If anyone has successfully set up an OS1-32 with FAST-LIO, I’d love to hear your insights.

Since I'm still new to this, any explanations or troubleshooting steps would help me a lot.

Thanks in advance!

giangthewalkingman, 2025-03-19:
same issue

Griwa88, 2025-03-24:
in preprocess.h under change ring from uint8_t to uint16_t.
From:
namespace ouster_ros {
  struct EIGEN_ALIGN16 Point {
      PCL_ADD_POINT4D;
      float intensity;
      uint32_t t;
      uint16_t reflectivity;
      uint8_t  ring;
      uint16_t ambient;
      uint32_t range;
      EIGEN_MAKE_ALIGNED_OPERATOR_NEW
  };
}  // namespace ouster_ros

// clang-format off
POINT_CLOUD_REGISTER_POINT_STRUCT(ouster_ros::Point,
    (float, x, x)
    (float, y, y)
    (float, z, z)
    (float, intensity, intensity)
    // use std::uint32_t to avoid conflicting with pcl::uint32_t
    (std::uint32_t, t, t)
    (std::uint16_t, reflectivity, reflectivity)
    (std::uint8_t, ring, ring)
    (std::uint16_t, ambient, ambient)
    (std::uint32_t, range, range)
)

To:
namespace ouster_ros {
  struct EIGEN_ALIGN16 Point {
      PCL_ADD_POINT4D;
      float intensity;
      uint32_t t;
      uint16_t reflectivity;
      uint16_t  ring;
      uint16_t ambient;
      uint32_t range;
      EIGEN_MAKE_ALIGNED_OPERATOR_NEW
  };
}  // namespace ouster_ros

// clang-format off
POINT_CLOUD_REGISTER_POINT_STRUCT(ouster_ros::Point,
    (float, x, x)
    (float, y, y)
    (float, z, z)
    (float, intensity, intensity)
    // use std::uint32_t to avoid conflicting with pcl::uint32_t
    (std::uint32_t, t, t)
    (std::uint16_t, reflectivity, reflectivity)
    (std::uint16_t, ring, ring)
    (std::uint16_t, ambient, ambient)
    (std::uint32_t, range, range)
)

closed 2025-03-25 by CrypticInfusion