hku-mars/FAST_LIO #410: Integration of Fast Lio with Hesai Lidar

CarstenKarl, 2025-05-09:
I am using the ROS2 branch of the Fast Lio. 

Going through the code I found that the code is mainly intended for the Livox Lidar. In fact, since I am not using the Livox lidar, the workspace is not building due to the fact that the Livox package is not installed.

What are the necessary steps to be taken/ the lines to be edited, to use the Hesai XT16 Lidar with the algorithm.

Thank you

marco-monforte, 2025-05-14:
Hi @CarstenKarl ,

I'm working with the same lidar on a Go2 robot and having the same issue. I had to install the Livox packages and create a customised config file for the Hesai XT16.

This is my config file, and I can launch the algorithm, but the odometry jumps, and the map is not built correctly. Hope it can be a helpful starting point for you, and we can get to the solution together:

```
/**:
    ros__parameters:
        feature_extract_enable: false
        point_filter_num: 1
        max_iteration: 3
        filter_size_surf: 0.5
        filter_size_map: 0.5
        cube_side_length: 1000.0
        runtime_pos_log_enable: false
        map_file_path: "./test.pcd"

        common:
            lid_topic:  "/lidar_points"
            imu_topic:  "/utlidar/imu"
            time_sync_en: false         # ONLY turn on when external time synchronization is really not possible
            time_offset_lidar_to_imu: 0.0 # Time offset between lidar and IMU calibrated by other algorithms, e.g. LI-Init (can be found in README).
                                        # This param will take effect no matter what time_sync_en is. So if the time offset is not known exactly, please set as 0.0

        preprocess:
            lidar_type: 2                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR, 
            scan_line: 32
            scan_rate: 10                # only need to be set for velodyne, unit: Hz,
            timestamp_unit: 2            # the unit of time/t field in the PointCloud2 rostopic: 0-second, 1-milisecond, 2-microsecond, 3-nanosecond.
            blind: 2.0

        mapping:
            acc_cov: 0.1
            gyr_cov: 0.1
            b_acc_cov: 0.0001
            b_gyr_cov: 0.0001
            fov_degree:    360.0
            det_range:     100.0
            extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic,
            extrinsic_T: [ 0.145, 0., 0.11]
            extrinsic_R: [ 1., 0., 0., 
                        0., 1., 0., 
                        0., 0., 1.]

        publish:
            path_en:  false
            scan_publish_en:  true       # false: close all the point cloud output
            dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
            scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

        pcd_save:
            pcd_save_en: true
            interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                        # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

closed 2025-06-06 by stale[bot]

CarstenKarl, 2025-06-17:
Hi @marco-monforte ,

Given that the Fast_Lio works with IMU data, what IMU are you using, please?

CarstenKarl, 2025-06-17:
Hi @marco-monforte,

Did you manage to solve the problem?

I am currently running the system on my laptop, initially I am able to see the map on RVIZ2 but after some time or when I move the IMU, multiples of the lidar points with offset are displayed and then no points are displayed. Then in the terminal it says "No effective points".

marco-monforte, 2025-06-17:
I'm having the same issue! And no, haven't been able to solve it.

It's like the map moves around and then jumps farther and farther away.

If I manage to solve it, I'll come back here. Please so the same.

CarstenKarl, 2025-06-17:
Thank you. 

I will keep you updated.

it5meyash, 2025-06-20:
Hello everyone Hope you all are doing well. Did anyone tried D-LIO SLAM on go2 i'm working with Unitree Go2 with Hesai XT-16

it5meyash, 2025-06-20:
Hi @marco-monforte,

Can we connect on the same topic over a Meet?

it5meyash, 2025-06-20:
Okay till that time I will find a solution.

Vincenzo-Saffioti, 2025-06-27:
@marco-monforte I'm also interested in joining the meeting 'cause i've the same problem in Go1 robot and Hesai xt16, keep me in mind

CarstenKarl, 2025-07-22:
Hi,

@marco-monforte  apologies, I was unavailable for a short while. Did you manage to solve the issue, please?

CarstenKarl, 2025-07-26:
I see. Yes, I did the same with the livox dependencies.

The Fast Lio2 built successfully and integrated the lidar and imu topics with it. However I constantly get:
-  The error message: [fastlio_mapping-1] Failed to find match for field 'time'.
- The location of the location starts stable then drifts away, even if nothing is moved
- the map is not being generated, as I cannot see it in rviz and when I send the /save_map server an error occurs because there is no data in the map.

@Vincenzo-Saffioti, any help would be appreciated
Thank you all

CarstenKarl, 2025-08-01:
Hi @marco-monforte ,

Would it be possible to share with me the Lio sam2 and the Hesai folders, such that I can compare with mine and see where I am wrong, please?

Thanks you

t300rsnecmi, 2025-08-13:
is there any update for support hesai ?

buaazyd, 2026-01-29:
Does anyone know how to solve this problem?