hku-mars/FAST_LIO #414: 3D map savig issue.

Kireiji02, 2025-06-17:
Hello, I am currently using topics published for a rosbag file. The map in RVIZ2 works perfectly but I couldn't save 3D map with PCD map saver (I'm pretty sure I tried every directories). I even tried to use ros2 service call but it doesn't seem to find any point clouds despite RVIZ2 mapping working fine.

[fastlio_mapping-1] [INFO] [1750139914.809540957] [laser_mapping]: Saving map to ./PCD/test.pcd...
[fastlio_mapping-1] terminate called after throwing an instance of 'pcl::IOException'
[fastlio_mapping-1]   what():  : [pcl::PCDWriter::writeBinary] Input point cloud has no data!

Thank you in advance!

AAArthurLiu, 2025-06-24:
Met with the exactly same issue

closed 2025-07-16 by stale[bot]

CramTenahc, 2025-07-31:
I also have this problem