hku-mars/FAST_LIO #416: The problem of uncertainty in code results

xiaotiantiian, 2025-06-19:
When I am running the same dataset, I output the timestamp, pos, and rot for each frame. At some point in the middle, the values start to differ from the previous dataset. With the same timestamp, the number of lidar points used in each frame, and the time between the first and last frames of the lidar,The timestamps of each IMU frame used were compared and all were normal. But there will still be different results, and this problem was discovered when looking at the final POS output. May I ask what may have caused it?

Ecstasy-EC, 2025-06-21:
Hi, thanks for the issue.  It is coming from the multi-threading in ikd-Tree. Specifically, the double-threaded rebuilding process might lead to different nearest neighbor search results (let's say, different points with the same distance). Therefore, the residuals will be different and the update in the pos will be different.

xiaotiantiian, 2025-06-23:
Thank you for your answer. Regarding the issue of using multithreading in ikdtree, we have disabled the use of multithreading in cmakelists and sharing models. However, the problem still exists. I have also output the timestamp, original point cloud, downsampling point cloud, and the number of point clouds in ikd_tree for each frame. Even with the same result (pos), where the timestamp, original point cloud, and downsampling point cloud for each frame are the same, the number of point clouds in ikd_tree for each frame will be different. May I ask if there are any other ideas? thank.

Ecstasy-EC, 2025-06-23:
> Thank you for your answer. Regarding the issue of using multithreading in ikdtree, we have disabled the use of multithreading in cmakelists and sharing models. However, the problem still exists. I [...]

Did u disable the second thread in ikdtree? From what u described it seems that u didn't do that. And you should not do that as it will heavily destroy the efficiency (peaks will appear when rebuilding large subtree)

Ecstasy-EC, 2025-06-23:
> Yes, I have disabled the add_definitions(-DMP_EN) in the CMakeLists file 
> 
> and commented out the following section in the h_share_model function:
> 
> // #ifdef MP_EN
> //     omp_set_num_threads(MP_PROC_NUM);
> // #pragma omp parallel for
> // #endif
> 
> I want to resolve the issue of non-deterministic computations in the code. My goal is to ensure that the same set of data produces consistent results across multiple runs, but this hasn't been achieved yet. I hope you can provide some insights.

Actually what you did here is not disabling the double threading in ikdtree but the multi threading for parallel search. If you have read the paper, u will notice that the double threading is for partial rebuilding. It is totally different from what you are doing now.

closed 2025-07-15 by stale[bot]