hku-mars/FAST_LIO #419: IMU drift and no map is saved – ROS 2 branch (Foxy)

lopf35, 2025-07-13:
Hi,
we are using the ROS 2 branch of FAST-LIO with ROS 2 Foxy on our Unitree G1 platform (running Ubuntu 20.04). We only installed the standard FAST-LIO ROS 2 branch (no localization modules yet).

We are experiencing two issues:

   1. Extreme IMU drift after a few minutes:
    Even if the robot is standing still and not moving at all, the body frame starts to drift extremely after around 2 minutes. This is not caused by movement or vibration, but purely time-dependent. At the beginning, the system works well – even when moving through space – but the drift starts regardless of whether the robot moves or not. Restarting the FAST-LIO node resets the drift temporarily.

  2.  No map saved automatically:
    Even after running for some time and moving through the environment, no .pcd map is saved anywhere. We are unsure if the current ROS 2 implementation supports this feature or if we missed some configuration.

Any help or clarification would be much appreciated. Thanks in advance!

NiklasVoigt, 2025-08-05:
problem is not solved

rFalque, 2025-08-14:
About the map, I got it by un-commentating the lines 516 to 543 in `src/laserMapping.cpp`. Got it saved in a different path than the `config.yaml` file. It went to `/home/user/ros2_ws/src/FAST_LIO/PCD/scans.pcd`.

I have also change `pcd_save_en` to `true` in line 76. Not sure if this is overwritten by the config file

inkccc, 2025-08-17:
我也遇到的第一个问题 我的ubunt是22.04.5 使用的是ros2humble 激光雷达是LivoxMid360 我不知道应该如何解决 启动fast_lio2时 即使雷达静止也会出现漂移现象 我一开始认为问题出现在livox_ros_driver2 我尝试使用PTP时间同步 发现/livox/lidar 和 /livox/imu即使在不适用时间同步的情况下 也不会出现离谱的时间差 但是fast_lio2的日志却经常输出两个话题之间有严重的时间不同步问题

closed 2025-09-27 by stale[bot]

xiaopengpeng, 2026-01-05:
> 我也遇到的第一个问题 我的ubunt是22.04.5 使用的是ros2humble 激光雷达是LivoxMid360 我不知道应该如何解决 启动fast_lio2时 即使雷达静止也会出现漂移现象 我一开始认为问题出现在livox_ros_driver2 我尝试使用PTP时间同步 发现/livox/lidar 和 /livox/imu即使在不适用时间同步的情况下 也不会出现离谱的时间差 但是fast_lio2的日志却经常输出两个话题之间有严重的时间不同步问题

@inkccc  你好，请问你解决了吗，我现在也遇到这个问题

xiaopengpeng, 2026-01-05:
> > > > > 我也遇到的第一个问题 我的ubunt是22.04.5 使用的是ros2humble 激光雷达是LivoxMid360 我不知道应该如何解决 启动fast_lio2时 即使雷达静止也会出现漂移现象 我一开始认为问题出现在livox_ros_driver2 我尝试使用PTP时间同步 发现/livox/lidar 和 /livox/imu即使在不适用时间同步的情况下 也不会出现离谱的时间差 但是fast_lio2的日志却经常输出两个话题之间有严重的时间不同步问题
> > > > 
> > > > 
> > > > [@inkccc](https://github.com/inkccc) 你好，请问你解决了吗，我现在也遇到这个问题
> > > 
> > > 
> > > 不清楚后面是怎么解决的 不过最近我没有使用fastlio做slam或里程计 而是使用pointlio 效果比fastlio会好一点 https://github.com/inkccc/small_point_lio
> > 
> > 
> > ok 后面我也尝试下，我最近测了很长时间，在最开始的一个小时很好，定位很准，导航也没问题。一个小时后就突然严重漂移。。
> 
> 我倒是没有测试过这么长时间 你有给雷达添加散热装置吗 运行太久温度会对imu产生影响 如果同时打开建图的话 需要查看内存是否充足

散热我倒是没有加，我用livox viewer观察过，好像刚开机和运行一段时间雷达温度都是60度左右。
内存也很充足，我也试了ptp之类的时间同步，都没有效果，
我试过将原始点云的发布频率提高一倍，这样能稳定运行2个小时多，但是这样点云数量又会少一倍，转成的2维雷达点太少，怕影响导航，简直要崩溃了，不知道该从哪下手，pointlio 这个您有试过长时间运行吗，效果可以吗

xiaopengpeng, 2026-01-05:
> > > > > > > 我也遇到的第一个问题 我的ubunt是22.04.5 使用的是ros2humble 激光雷达是LivoxMid360 我不知道应该如何解决 启动fast_lio2时 即使雷达静止也会出现漂移现象 我一开始认为问题出现在livox_ros_driver2 我尝试使用PTP时间同步 发现/livox/lidar 和 /livox/imu即使在不适用时间同步的情况下 也不会出现离谱的时间差 但是fast_lio2的日志却经常输出两个话题之间有严重的时间不同步问题
> > > > > > 
> > > > > > 
> > > > > > [@inkccc](https://github.com/inkccc) 你好，请问你解决了吗，我现在也遇到这个问题
> > > > > 
> > > > > 
> > > > > 不清楚后面是怎么解决的 不过最近我没有使用fastlio做slam或里程计 而是使用pointlio 效果比fastlio会好一点 https://github.com/inkccc/small_point_lio
> > > > 
> > > > 
> > > > ok 后面我也尝试下，我最近测了很长时间，在最开始的一个小时很好，定位很准，导航也没问题。一个小时后就突然严重漂移。。
> > > 
> > > 
> > > 我倒是没有测试过这么长时间 你有给雷达添加散热装置吗 运行太久温度会对imu产生影响 如果同时打开建图的话 需要查看内存是否充足
> > 
> > 
> > 散热我倒是没有加，我用livox viewer观察过，好像刚开机和运行一段时间雷达温度都是60度左右。 内存也很充足，我也试了ptp之类的时间同步，都没有效果， 我试过将原始点云的发布频率提高一倍，这样能稳定运行2个小时多，但是这样点云数量又会少一倍，转成的2维雷达点太少，怕影响导航，简直要崩溃了，不知道该从哪下手，pointlio 这个您有试过长时间运行吗，效果可以吗
> 
> pointlio我没有长时间运行过 不过里程计还算稳定 我没有长时间运行的需求 既然有长时间的需求 imu累积会有误差 可以在运行一定时间后保存里程计 重启继续运行会不会能解决这个问题

嗯，再折腾两天，是在不行就只能软重启了

stz2233, 2026-09-04:
> 嗯，再折腾两天，是在不行就只能软重启了

@xiaopengpeng 你好，请问你解决了吗